guaix-ucm/megaradrp #137: New output line for the "Fine acquisition with the LCB" recipe

gildepaz, 2017-07-28:
The output of the "Fine acquisition with the LCB" recipe should include (in addition to the information already provided) the offset that should be applied to the telescope to properly acquire the target in the center of the LCB in arcsec. Examples:
- For current offset = [+0.4,0] it should say "Offset telescope by 0.4" W, 0.0" N
- For current offset = [-0.4,0] it should say "Offset telescope by 0.4" E, 0.0" N
- For current offset = [0,+0.4] it should say "Offset telescope by 0.0" E, 0.4" N
- For current offset = [0,-0.4] it should say "Offset telescope by 0.0" E, 0.4" S

sergiopasra, 2017-07-28:
This depends on the instrument angle, IPA, etc do you have the equations?

gildepaz, 2018-02-01:
I think these are the ones you use for the DFAgent.

closed 2018-07-26 by gildepaz