guaix-ucm/megaradrp #163: Model Map not created

pgomezalvarez, 2017-12-05:
When trying to generate MEGARA Model Map:
`numina run 3_MegaraModelMap.yaml -r ../control.yaml `
With the following YAML

```
id: 3_LR_V_ModelMap
mode: MegaraModelMap
instrument: MEGARA
frames:
- 0001312112-20170830-MEGARA-MegaraLcbImage.fits
- 0001312113-20170830-MEGARA-MegaraLcbImage.fits
- 0001312114-20170830-MEGARA-MegaraLcbImage.fits
```

After some time... the process finished but no result:

[image]

The console output was:

```
(megara2_7) Pedro-MBP-2:tutorial pgomez$ numina run 3_MegaraModelMap.yaml -r ../control.yaml 

DEBUG: Numina simple recipe runner version 0.15.dev5
INFO: Loading observation results from '3_MegaraModelMap.yaml'
DEBUG: load observation result with id 3_LR_V_ModelMap
INFO: reading control from ../control.yaml
INFO: control format version 1
DEBUG: pipeline from CLI is 'default'
DEBUG: recipe class is <class 'megaradrp.recipes.calibration.modelmap.ModelMapRecipe'>
DEBUG: enable intermediate results
DEBUG: update recipe runinfo
DEBUG: recipe created
numina.dal.diskfiledal INFO search master_slitflat of type MasterSlitFlat()
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/result.json
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
numina.dal.diskfiledal INFO found /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
numina.dal.diskfiledal INFO search master_bias of type MasterBias()
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/result.json
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/.DS_Store
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/00_master_bias.fits
numina.dal.diskfiledal INFO found /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/00_master_bias.fits
/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/00_master_bias.fits
numina.dal.diskfiledal INFO search master_dark of type MasterDark()
numina.dal.diskfiledal INFO type MasterDark compatible with tags {'insmode': 'LCB', 'vph': 'LR-V'} not found
numina.dal.diskfiledal INFO search master_bpm of type MasterBPM()
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/result.json
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
numina.dal.diskfiledal INFO found /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
numina.dal.diskfiledal INFO search master_traces of type TraceMap()
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-V/result.json
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-V/.DS_Store
loadpath= /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-V/00_master_traces.json
numina.dal.diskfiledal INFO found /Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-V/00_master_traces.json
/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-V/00_master_traces.json
DEBUG: recipe input created
DEBUG: recipe requires 'processes', value is 0
DEBUG: recipe requires 'master_slitflat', value is DataFrame(filename=u'/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits')
DEBUG: recipe requires 'debug_plot', value is 0
DEBUG: recipe requires 'master_bias', value is DataFrame(filename=u'/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/00_master_bias.fits')
DEBUG: recipe requires 'master_dark', value is None
DEBUG: recipe requires 'master_bpm', value is DataFrame(filename=u'/Users/pgomez/Fractal/Megara/MegaraCommissionning/dp/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits')
DEBUG: recipe requires 'obresult', value is ObservationResult(id=3_LR_V_ModelMap, instrument=MEGARA, mode=MegaraModelMap)
DEBUG: recipe requires 'master_traces', value is TraceMap(instrument=MEGARA, uuid=8ca7f7b3-d8f5-11e7-9944-c4b301d558b3)
DEBUG: recipe provides ProcessedRSS, 
DEBUG: recipe provides QualityControlProduct, 
DEBUG: recipe provides ModelMap, 
DEBUG: recipe provides ProcessedImage, 
[... 187 lines cut ...]
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 59, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 197, in mode_run_common_obs
    workenv=workenv, task_control=task_control)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 232, in run_recipe
    completed_task = run_recipe_timed(recipe, rinput, task)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 246, in run_recipe_timed
    result = recipe(rinput)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/core/recipes.py", line 156, in __call__
    result = self.run(recipe_input)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/megaradrp-0.6.dev4-py2.7.egg/megaradrp/recipes/calibration/modelmap.py", line 193, in run
    reduced_rss = calibrator_aper(reduced_copy)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/flow/processing.py", line 82, in __call__
    img = self.run(img)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/megaradrp-0.6.dev4-py2.7.egg/megaradrp/processing/aperture.py", line 59, in run
    rssdata = self.tracemap.aper_extract(img[0].data, processes=self.processes)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/megaradrp-0.6.dev4-py2.7.egg/megaradrp/products/modelmap.py", line 102, in aper_extract
    return aper_extract(self, self._wcols, img)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/megaradrp-0.6.dev4-py2.7.egg/megaradrp/products/modelmap.py", line 267, in aper_extract
    res = lsqr(val, yl)
  File "/Users/pgomez/opt/anaconda/envs/megara2_7/lib/python2.7/site-packages/scipy/sparse/linalg/isolve/lsqr.py", line 369, in lsqr
    cs1 = rhobar / rhobar1
ZeroDivisionError: float division by zero
(megara2_7) Pedro-MBP-2:tutorial pgomez$ 
```

sergiopasra, 2017-12-05:
The message:

`
The exact solution is  x = 0
`

before the crash is very curious. 

I have tries to reproduce this but in my case, the recipe goes to the end.

What versions of scipy and numpy do you have?

pgomezalvarez, 2017-12-05:
This is wath I get from the console in MEGARA environment:
```
(megara2_7) Pedro-MBP-2:tutorial pgomez$ python
Python 2.7.13 |Continuum Analytics, Inc.| (default, Dec 20 2016, 23:05:08) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://anaconda.org
>>> import numpy
>>> numpy.version.version
'1.13.0'
>>> import scipy
>>> scipy.version.version
'0.19.0'
>>> 
```

pgomezalvarez, 2017-12-06:
I reinstalled numina and megaradrp alltogether. Now the packages are:
>>> import numpy
>>> numpy.version.version
'1.13.3'
>>> import scipy
>>> scipy.version.version
'1.0.0'

But the error stands...

pgomezalvarez, 2017-12-06:
I Installed the conda from zero, created a new environment with Python3 and compiled the numina and megaradrp packages from scratch. Now the problem is gone :-)
The list of dependencies in this new environment is:

```
Pedro-MacBook-Pro-2:tutorial pgomez$ source activate megara3
(megara3) Pedro-MacBook-Pro-2:tutorial pgomez$ conda list
# packages in environment at /Users/pgomez/opt/anaconda/envs/megara3:
#
astropy                   2.0.2                     <pip>
attrs                     17.3.0                    <pip>
ca-certificates           2017.08.26           ha1e5d58_0  
certifi                   2017.11.5        py36ha569be9_0  
cycler                    0.10.0           py36hfc81398_0  
decorator                 4.1.2                     <pip>
freetype                  2.8                  h12048fb_1  
intel-openmp              2018.0.0             h8158457_8  
libcxx                    4.0.1                h579ed51_0  
libcxxabi                 4.0.1                hebd6815_0  
libedit                   3.1                  hb4e282d_0  
libffi                    3.2.1                h475c297_4  
libgfortran               3.0.1                h93005f0_2  
libpng                    1.6.32               hd1e8b91_4  
matplotlib                2.1.0                     <pip>
matplotlib                2.1.0            py36h5068139_0  
megaradrp                 0.6.dev4                  <pip>
mkl                       2018.0.1             hfbd8650_4  
ncurses                   6.0                  hd04f020_2  
networkx                  2.0                       <pip>
numina                    0.15.dev5                 <pip>
numpy                     1.13.3           py36h2cdce51_0  
openssl                   1.0.2m               h86d3e6a_1  
pip                       9.0.1            py36h1555ced_4  
pluggy                    0.6.0                     <pip>
py                        1.5.2                     <pip>
pyparsing                 2.2.0            py36hb281f35_0  
pytest                    3.3.1                     <pip>
python                    3.6.3                h5ce8c04_4  
python-dateutil           2.6.1            py36h86d2abb_1  
pytz                      2017.3           py36hf0bf824_0  
PyWavelets                0.5.2                     <pip>
readline                  7.0                  hc1231fa_4  
scikit-image              0.13.1                    <pip>
scipy                     1.0.0            py36h1de22e9_0  
setuptools                36.5.0           py36h2134326_0  
six                       1.11.0                    <pip>
six                       1.11.0           py36h0e22d5e_1  
sqlite                    3.20.1               h7e4c145_2  
tk                        8.6.7                h35a86e2_3  
tornado                   4.5.2            py36h468dda9_0  
wheel                     0.30.0           py36h5eb2c71_1  
xz                        5.2.3                h0278029_2  
zlib                      1.2.11               hf3cbc9b_2  
```

sergiopasra, 2017-12-12:
It looks like a problem with some numerical library. Please, reopen if  you encounter this problem again

pgomezalvarez, 2017-12-13:
It seems that also Ana was also affected. Could be related with a recent update I did to OsX High Sierra?

gildepaz, 2018-01-29:
Same problem with me. I will be trying to fix it following Pedro's way!

gildepaz, 2018-01-30:
I tried doing:
conda update conda
conda update scipy
conda update numpy
That put all the packages within the megara environment to versions equal or even more recent than those of Pedro but still the MegaraModelMap does not work (I still get "ZeroDivisionError: float division by zero" error"). 

I have also then tried:
conda remove --name megara --all
conda install -c sergiopasra/label/dev megaradrp
source activate megara
but it does not change any package, so it won't work (although I have executed just in case)

gildepaz, 2018-01-31:
Indeed, it does not work yet. I will try with a deeper update (everything from scratch)

gildepaz, 2018-01-31:
I finally made it work by installing anaconda3 (i.e. anaconda for Python3). In order to do that without removing the previous version (anaconda2) you can simply go to https://www.anaconda.com/download/ and install the anaconda for Python3. That would automatically change your .bash_profile to 
export PATH="$HOME/anaconda3/bin:$PATH" (instead of export PATH="$HOME/anaconda2/bin:$PATH"). If you add $HOME/anaconda3/bin to the "setenv PATH" of you .tcshrc you could also run "numina" within tcsh without any environment. 
Alternatively, the recommended option, is to create again the environment by doing "conda create --name megara" and then "source activate megara" in bash. 

You do not have to reinstall numina or megaradrp you can simply go to your github directory and do:
cd numina; git pull; python setup.py build; python setup.py install
cd ../megaradrp; git pull; python setup.py build; python setup.py install

That should be enough. So, in summary, if you are using a Mac, consider using the distribution for Python3. That should be included in the documentation / cookbook.

closed 2018-01-31 by gildepaz