guaix-ucm/megaradrp #168: having problems running the MEGARA task FiberFlat

BillyTastroD, 2017-12-20:
Dear Sergio and others,

Please have a look at the error messages, i.e., traceback here. This is caused possibly due to a typo, a missing 'h' in the term 'smooting_window.

Traceback (most recent call last):
  File "/Users/bililigndullo/anaconda3/envs/megara/bin/numina", line 11, in <module>
    load_entry_point('numina==0.15.dev5', 'console_scripts', 'numina')()
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/cli.py", line 151, in main
    args.command(args, extra_args)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/clirun.py", line 108, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 59, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 197, in mode_run_common_obs
    workenv=workenv, task_control=task_control)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 232, in run_recipe
    completed_task = run_recipe_timed(recipe, rinput, task)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 246, in run_recipe_timed
    result = recipe(rinput)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-macosx-10.7-x86_64.egg/numina/core/recipes.py", line 156, in __call__
    result = self.run(recipe_input)
  File "/Users/bililigndullo/anaconda3/envs/megara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/recipes/calibration/flat.py", line 193, in run
    rss_wl2 = self.obtain_fiber_flat(rss_wl, rinput.master_wlcalib, window=rinput.smooting_window)
AttributeError: 'FiberFlatRecipeInput' object has no attribute 'smooting_window'

sergiopasra, 2017-12-20:
> Assigned #168 <https://github.com/guaix-ucm/megaradrp/issues/168> to
> @sergiopasra <https://github.com/sergiopasra>.

Please update, I think I fixed this yesterday

sergiopasra, 2017-12-20:
Yes, it was #167

BillyTastroD, 2017-12-20:
I am just seeing issue #167 now since I was not in that other mail thread and did not receive notifications about it. Having git pull just now and running the FiberFlat task in a new terminal window the problem still persists.

closed 2017-12-20 by sergiopasra

BillyTastroD, 2017-12-20:
Thanks Sergio!