guaix-ucm/megaradrp #174: Installation instructions in readthedocs poorly structured and possibly out of date

gildepaz, 2018-01-13:
In http://megara-drp.readthedocs.io/en/latest/installation.html the information on how to install the MEGARA DRP is poorly structured and possibly out of date. In particular:
- The use of python 3 is not considered but the DRP is currently able to run under python 3
- Pip is recommended as a possible way to install the DRP but this is only recommended for non-Mac users. There should be some platform specific instructions for the installation. 
- Menu items "Using Conda", "Using Pip" and "Building from source" are at the same menu level as "Checking the installation" or "Development version". There should be an independent menu level for "Installing the code" itself where "Using Conda", "Using Pip" and "Building from source" would be included separated from "Checking the installation" or "Development version".
- It is not clear how critical the "Development version" version should be. 
- Installing the MEGARA DFP should be in a completely different menu level so regular users would not see unless explicit clicked on that menu.

gildepaz, 2018-01-31:
The documentation should mention that for a Mac the Conda for Python3 is recommended.

sergiopasra, 2018-01-31:
Python 3 is recommended for everyone.

closed 2026-10-14 by sergiopasra