guaix-ucm/megaradrp #176: Standard star AB mag file should be taken from LcbStdStar.yaml file not from control.yaml

gildepaz, 2018-01-16:
Right now, the name of the file that has the standard star flux calibration should be provided in the control.yaml file. Since the name of this file should change from observation to observation I do not think that that parameter should be modified in the control.yaml file but in the specific LcbStdStar.yaml file being used, e.g. as a requirement.

sergiopasra, 2018-01-16:
> Assigned #176 <https://github.com/guaix-ucm/megaradrp/issues/176> to
> @sergiopasra <https://github.com/sergiopasra>.

This should actually work, by writing the name in a section called
requirements in the OB file. You have to write the values like this:

requirements:
  reference_star: filename.txt

gildepaz, 2018-01-16:
It works if you use ... 
 reference_spectrum: filename.txt
instead of ... 
 reference_star: filename.txt 
(leading space is critical, btw) Thx!

gildepaz, 2018-01-17:
It does run at the beginning but finally I get the following error (the same file has been used successfully including it in the control.yaml file. I have tried both putting the name of the file with and without commas.

gildepaz, 2018-01-17:
Aunque corre, cuando llega al siguiente punto casca con el "reference_spectrum":

megaradrp.processing.fiberflat DEBUG correct fiberflat in image uuid:cff3d254-fae5-11e7-b523-a20090926401
numina.recipes.megara INFO start sky subtraction
numina.recipes.megara INFO obtain fiber information
numina.recipes.megara DEBUG sky fibers are: %s
numina.recipes.megara INFO end sky subtraction
numina.recipes.megara DEBUG adding 3 nrings
numina.recipes.megara DEBUG adding 37 fibers
numina.recipes.megara INFO extracting star
numina.recipes.megara DEBUG Configuration UUID is b7dcd9d1-0b60-4b43-b26e-d2c9868d5e20
numina.recipes.megara INFO Using 37 nearest fibers
numina.recipes.megara INFO For point [0.49, -0.3]
Traceback (most recent call last):
  File "/Users/agpaz/anaconda2/envs/megara/bin/numina", line 11, in <module>
    load_entry_point('numina==0.15.dev5', 'console_scripts', 'numina')()
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/cli.py", line 154, in main
    args.command(args, extra_args)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirun.py", line 108, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 59, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 197, in mode_run_common_obs
    workenv=workenv, task_control=task_control)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 232, in run_recipe
    completed_task = run_recipe_timed(recipe, rinput, task)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 246, in run_recipe_timed
    result = recipe(rinput)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/numina-0.15.dev5-py2.7-macosx-10.7-x86_64.egg/numina/core/recipes.py", line 156, in __call__
    result = self.run(recipe_input)
  File "/Users/agpaz/anaconda2/envs/megara/lib/python2.7/site-packages/megaradrp-0.6.dev4-py2.7.egg/megaradrp/recipes/calibration/lcbstdstar.py", line 101, in run
    star_interp = interp1d(rinput.reference_spectrum[:,0], rinput.reference_spectrum[:,1])
IndexError: too many indices for array

closed 2018-01-19 by sergiopasra