guaix-ucm/megaradrp #180: error in 'MegaraTwilightFlatImage'

ccatalan, 2018-01-23:
File "/usr/lib64/python3.6/site-packages/matplotlib/backends/__init__.py", line 32, in pylab_setup
    globals(),locals(),[backend_name],0)
ModuleNotFoundError: No module named 'matplotlib.backends.backend_qt5agg'

Thanks!

sergiopasra, 2018-01-23:
Uhmmm, this belongs to @nicocardiel

nicocardiel, 2018-01-23:
It seems that python3.6 requires the package pyside2 in order to work properly with the Qt5 backend. It is possible to install pyside2 using conda:
`% conda install -c conda-forge pyside2`
Try doing this and let me know if it works.

sergiopasra, 2018-01-23:
We have installed a rpm package instead, as @ccatalan is using virtualenv and not conda

sergiopasra, 2018-01-23:
The DRP shouln't depend on a particular matplotlib backend. It's OK if utility scripts use this but not if it appears in the recipes. We need to check the imports to see where this comes

sergiopasra, 2018-01-23:
@ccatalan Could you paste a little bit more of the backtrace? (you could uninstall the rpm package, run the recipe and then install it again)

ccatalan, 2018-01-23:
Traceback (most recent call last):
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/bin/numina", line 11, in <module>
    load_entry_point('numina==0.15.dev5', 'console_scripts', 'numina')()
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/user/cli.py", line 154, in main
    args.command(args, extra_args)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/user/clirun.py", line 108, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 59, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 128, in mode_run_common_obs
    recipe = dal.search_recipe_from_ob(obsres)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/dal/absdal.py", line 75, in search_recipe_from_ob
    return self.search_recipe(instrument, mode, pipeline)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/dal/absdal.py", line 57, in search_recipe
    recipe = this_pipeline.get_recipe_object(mode)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/core/pipeline.py", line 80, in get_recipe_object
    recipe = self._get_base_object(recipe_entry)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/core/pipeline.py", line 60, in _get_base_object
    Cls = self._get_base_class(entry)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/core/pipeline.py", line 56, in _get_base_class
    return numina.core.objimport.import_object(recipe_fqn)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/core/objimport.py", line 35, in import_object
    mm = importlib.import_module(mods)
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/recipes/calibration/twilight.py", line 22, in <module>
    import megaradrp.requirements as reqs
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/requirements.py", line 16, in <module>
    import megaradrp.products
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/products/__init__.py", line 7, in <module>
    from .wavecalibration import WavelengthCalibration
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/products/wavecalibration.py", line 13, in <module>
    from numina.array.wavecalib.arccalibration import SolutionArcCalibration
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/array/wavecalib/arccalibration.py", line 32, in <module>
    from ..display.pause_debugplot import pause_debugplot
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/array/display/pause_debugplot.py", line 23, in <module>
    from .matplotlib_qt import plt
  File "/home/ccatalan/MEGARA_all/DRP/envmegara/lib/python3.6/site-packages/numina-0.15.dev5-py3.6-linux-x86_64.egg/numina/array/display/matplotlib_qt.py", line 3, in <module>
    import matplotlib.pyplot as plt
  File "/usr/lib64/python3.6/site-packages/matplotlib/pyplot.py", line 115, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/usr/lib64/python3.6/site-packages/matplotlib/backends/__init__.py", line 32, in pylab_setup
    globals(),locals(),[backend_name],0)
ModuleNotFoundError: No module named 'matplotlib.backends.backend_qt5agg'

nicocardiel, 2018-01-25:
I have removed the explicit use of the Qt5 backend. This should fix the problem.

ccatalan, 2018-01-25:
I have removed the qt5 package in Fedora and thanks to the new update made by @nicocardiel the 'MegaraTwilightFlatImage' package is working properly!

nicocardiel, 2018-01-25:
Good job team!

closed 2018-01-25 by nicocardiel