guaix-ucm/megaradrp #185: Problem with astrometry in header

acasmor, 2018-02-02:
I have a target with a IPA= 196.744 degrees, so the N-E orientation should be as usual.
I´ve checked the transformation matrix in this case is wrong, in the header is written:

PC1_1   =   0.8217066233582691 / Coordinate transformation matrix element       
PC1_2   =    0.569910716804967 / Coordinate transformation matrix element       
PC1_3   =                  0.0 / Coordinate transformation matrix element       
PC2_1   =   -0.569910716804967 / Coordinate transformation matrix element       
PC2_2   =   0.8217066233582691 / Coordinate transformation matrix element       
PC2_3   =                  0.0 / Coordinate transformation matrix element       
PC3_1   =                  0.0 / Coordinate transformation matrix element       
PC3_2   =                  0.0 / Coordinate transformation matrix element       
PC3_3   =                  1.0 / Coordinate transformation matrix element  

instead of pc1_1 and pc2_2 equal to 1. and a diagonal matrix.
The keyword cdelt1 should be negative as well (for good orientation) and it is positive.
I´ve checked that doing those changes the cube of the target is well correlated with the hst image except for a offset of 2-3 arcsecs in x, y.

sergiopasra, 2018-02-16:
The sign of cdelt1 is fixed in the latest cube.py 

The angles are clearly wrong. This is something we should check in the code of the Agent. For the moment, cube.py recomputes the angles from IPA

sergiopasra, 2018-03-02:
With c148084 the PA angle and the PC matrix can be computed from IPA. Only the problem in the Agent remains

closed 2026-10-14 by sergiopasra