guaix-ucm/megaradrp #210: Sky bundle rejection

BillyTastroD, 2018-05-11:
Dear Sergio,

Please have a look this error. 

numina.dal.diskfiledal INFO found /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterFiberFlat/LCB/LR-U/master_fiberflat.fits
/Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterFiberFlat/LCB/LR-U/master_fiberflat.fits
numina.dal.diskfiledal INFO search master_twilight of type MasterTwilightFlat()
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterTwilightFlat/LCB/LR-U/result.json
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterTwilightFlat/LCB/LR-U/.DS_Store
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterTwilightFlat/LCB/LR-U/master_twilightflat.fits
numina.dal.diskfiledal INFO found /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterTwilightFlat/LCB/LR-U/master_twilightflat.fits
/Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterTwilightFlat/LCB/LR-U/master_twilightflat.fits
numina.dal.diskfiledal INFO search master_traces of type ModelMap()
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/ModelMap/LCB/LR-U/result.json
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/ModelMap/LCB/LR-U/.DS_Store
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/ModelMap/LCB/LR-U/master_model.json
numina.dal.diskfiledal INFO found /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/ModelMap/LCB/LR-U/master_model.json
/Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/ModelMap/LCB/LR-U/master_model.json
numina.dal.diskfiledal INFO search master_traces of type TraceMap()
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-U/result.json
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-U/.DS_Store
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-U/master_traces.json
numina.dal.diskfiledal INFO found /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-U/master_traces.json
/Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/TraceMap/LCB/LR-U/master_traces.json
numina.dal.diskfiledal INFO search master_sensitivity of type MasterSensitivity()
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSensitivity/LCB/LR-U/result.json
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSensitivity/LCB/LR-U/.DS_Store
loadpath= /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSensitivity/LCB/LR-U/master_sensitivity.fits
numina.dal.diskfiledal INFO found /Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSensitivity/LCB/LR-U/master_sensitivity.fits
/Users/bililigndullo/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSensitivity/LCB/LR-U/master_sensitivity.fits
Traceback (most recent call last):
  File "/Users/bililigndullo/anaconda3/envs/megara_S/bin/numina", line 11, in <module>
    sys.exit(main())
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/user/cli.py", line 144, in main
    args.command(args, extra_args)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/user/clirun.py", line 98, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/user/clirundal.py", line 44, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/user/clirundal.py", line 132, in mode_run_common_obs
    rinput = recipe.build_recipe_input(obsres, dal)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/core/recipes.py", line 239, in build_recipe_input
    return self.create_input(**result)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/core/recipes.py", line 108, in create_input
    return cls.RecipeInput(*args, **kwds)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/core/recipeinout.py", line 30, in __init__
    setattr(self, key, kwds[key])
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/core/dataholders.py", line 60, in __set__
    cval = self.convert(value)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/core/dataholders.py", line 335, in convert
    pre = self.type.convert(val)
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/types/datatype.py", line 165, in convert
    result = [self.internal.convert(o) for o in obj]
  File "/Users/bililigndullo/anaconda3/envs/megara_S/lib/python3.6/site-packages/numina/types/datatype.py", line 165, in <listcomp>
    result = [self.internal.convert(o) for o in obj]
AttributeError: 'NoneType' object has no attribute 'convert'

sergiopasra, 2018-05-14:
This was a numina problem, it's fixed in main development and also in the last bugfix release, so please
install numina 0.16.1

sergiopasra, 2018-05-16:
It's numina 0.16.2, sorry

closed 2018-05-16 by sergiopasra