guaix-ucm/megaradrp #213: Error running LcbStdStar: TypeError: 'slice' object is not iterable

gildepaz, 2018-06-19:
Traceback (most recent call last):
  File "/home/marxacaz/anaconda3/envs/megara/bin/numina", line 11, in <module>
    load_entry_point('numina==0.17.dev1', 'console_scripts', 'numina')()
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/cli.py", line 144, in main
    args.command(args, extra_args)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirun.py", line 98, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 44, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 182, in mode_run_common_obs
    workenv=workenv, task_control=task_control)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 216, in run_recipe
    completed_task = run_recipe_timed(recipe, rinput, task)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 230, in run_recipe_timed
    result = recipe(rinput)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-linux-x86_64.egg/numina/core/recipes.py", line 146, in __call__
    result = self.run(recipe_input)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/recipes/calibration/lcbstdstar.py", line 120, in run
    sens = generate_sensitivity(final, spectrum, star_interp, extinc_interp, wl_cover1, wl_cover2, sigma)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/megaradrp-0.6.dev4-py3.6.egg/megaradrp/processing/extractobj.py", line 329, in generate_sensitivity
    pixm1, pixm2 = wl_coverage1
TypeError: 'slice' object is not iterable

gildepaz, 2018-06-19:
The reported error was in vortex (linux Fedora) but it does not happen to me (Macbook Pro) but with a different dataset

gildepaz, 2018-06-19:
It could be due to a problem in the wavelength calibration file since we could not wavelength calibrate these data (see error message below) and had to use an old wavelength calibration file;

Traceback (most recent call last):
  File "/home/marxacaz/anaconda3/envs/megara/bin/numina", line 11, in <module>
    load_entry_point('numina==0.17.dev1', 'console_scripts', 'numina')()
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/cli.py", line 144, in main
    args.command(args, extra_args)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirun.py", line 98, in mode_run_ob
smode
    mode_run_common(args, extra_args, mode='obs')
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 44, in mode_run
_common
    return mode_run_common_obs(args, extra_args)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 182, in mode_ru
n_common_obs
    workenv=workenv, task_control=task_control)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 216, in run_rec
ipe
    completed_task = run_recipe_timed(recipe, rinput, task)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/user/clirundal.py", line 230, in run_rec
ipe_timed
    result = recipe(rinput)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/numina-
0.17.dev1-py3.6-linux-x86_64.egg/numina/core/recipes.py", line 146, in __call__
    result = self.run(recipe_input)
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/megarad
rp-0.6.dev4-py3.6.egg/megaradrp/recipes/calibration/arc.py", line 183, in run
    debugplot=debugplot
  File "/home/marxacaz/anaconda3/envs/megara/lib/python3.6/site-packages/megarad
rp-0.6.dev4-py3.6.egg/megaradrp/recipes/calibration/arc.py", line 257, in calibr
ate_wl
    wv_master_all = lines_catalog[:, 0]
IndexError: too many indices for array

sergiopasra, 2018-06-19:
Is this a new problem?

gildepaz, 2018-06-19:
Yes. It is a problem for wavelength calibration but it might come from a wrong format on the control.yaml file (Mario, please, be sure that the format for arc lines in control.yaml is the correct one)

sergiopasra, 2018-06-19:
I'm closing this. It there is an error in the code, please open a new issue

commit e21cc79a (2018-08-06): Obtain values from slice, not tuple (fixes #213)
closed 2018-06-19 by sergiopasra