guaix-ucm/megaradrp #216: Requirement: \\ master_traces: <file.json> not working

gildepaz, 2018-07-03:
While testing the use of "requirements:" in the recipes for feeding specific calibrations to different blocks with equal mode and VPH the requirement "master_traces:" does not work. It does not matter if the file in the data/ folder is a TraceMap or a ModelMap, both failed. The lines in the .yaml file (FiberFlat in my case) were: 
requirements:
 master_traces: master_traces_HR-R_26_6.json
OR
requirements:
 master_traces: master_model_HR-R_26_6.json

However, if I use the "requirements:" for the MasterBias by doing:
requirements:
 master_bias: master_bias_test.fits
IT WORKS!

The error for the "master_traces:" requirement is:

DEBUG: found insconf config uuid=ca3558e3-e50d-4bbc-86bd-da50a0998a48
DEBUG: pipeline from CLI is 'default'
DEBUG: recipe class is <class 'megaradrp.recipes.calibration.flat.FiberFlatRecipe'>
DEBUG: enable intermediate results
DEBUG: update recipe runinfo
DEBUG: recipe created
numina.recipes.megara DEBUG Use mode tagger to fill tags in OB
numina.recipes.megara DEBUG getting query fields per REQ
numina.recipes.megara DEBUG obresult has these query fields set()
numina.recipes.megara DEBUG master_bias has these query fields set()
numina.recipes.megara DEBUG master_dark has these query fields set()
numina.recipes.megara DEBUG master_bpm has these query fields set()
numina.recipes.megara DEBUG master_slitflat has these query fields set()
numina.recipes.megara DEBUG master_traces has these query fields []
numina.recipes.megara DEBUG smoothing_window has these query fields set()
numina.recipes.megara DEBUG extraction_offset has these query fields set()
numina.recipes.megara DEBUG master_wlcalib has these query fields set()
numina.dal.diskfiledal INFO search master_bias of type MasterBias()
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/result.json
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBias/original
numina.dal.diskfiledal INFO type MasterBias compatible with tags {'vph': 'HR-R', 'insmode': 'LCB'} not found
numina.dal.diskfiledal INFO search master_dark of type MasterDark()
numina.dal.diskfiledal INFO type MasterDark compatible with tags {'vph': 'HR-R', 'insmode': 'LCB'} not found
numina.dal.diskfiledal INFO search master_bpm of type MasterBPM()
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/result.json
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
numina.dal.diskfiledal INFO found /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
/Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterBPM/master_bpm.fits
numina.dal.diskfiledal INFO search master_slitflat of type MasterSlitFlat()
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/result.json
loadpath= /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
numina.dal.diskfiledal INFO found /Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
/Users/agpaz/Desktop/MEGARA/ca3558e3-e50d-4bbc-86bd-da50a0998a48/MasterSlitFlat/master_sflat.fits
Traceback (most recent call last):
  File "/Users/agpaz/anaconda3/bin/numina", line 11, in <module>
    load_entry_point('numina==0.17.dev1', 'console_scripts', 'numina')()
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/user/cli.py", line 144, in main
    args.command(args, extra_args)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/user/clirun.py", line 98, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 44, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 132, in mode_run_common_obs
    rinput = recipe.build_recipe_input(obsres, dal)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/core/recipes.py", line 251, in build_recipe_input
    result[key] = req.query(dal, obsres, options=query_option)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/core/dataholders.py", line 208, in query
    val = self.type.query(self.dest, dal, obsres, options=self.query_opts)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/types/multitype.py", line 36, in query
    result = subtype.query(name, dal, ob, options=options)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/types/base.py", line 74, in query
    return self.query_on_ob(name, obsres)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/types/base.py", line 98, in query_on_ob
    value = self._datatype_load(content)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/numina-0.17.dev1-py3.6-macosx-10.7-x86_64.egg/numina/types/structured.py", line 115, in _datatype_load
    result.__setstate__(state=state)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/megaradrp/products/modelmap.py", line 89, in __setstate__
    m.__setstate__(trace)
  File "/Users/agpaz/anaconda3/lib/python3.6/site-packages/megaradrp/products/modelmap.py", line 51, in __setstate__
    self._set_model(state['model'])
KeyError: 'model'

sergiopasra, 2018-08-05:
This should work with the latest commit in numina: https://github.com/guaix-ucm/numina/commit/bf527327e97c9ffe725c25698a10d19756d312bc 

Please test!

sergiopasra, 2018-08-06:
Reopen if current fix does not work

closed 2018-08-06 by sergiopasra