guaix-ucm/megaradrp #218: insert_fibers_in_extension.py not working under "megara" conda environment (under py3.6?)

gildepaz, 2018-07-30:
The insert_fibers_in_extension.py program (added as an attachment to this issue; renamed as .txt extension for github to allow me attach it
[insert_fibers_in_extension.txt](https://github.com/guaix-ucm/megaradrp/files/2240522/insert_fibers_in_extension.txt)
)
works when used outside the "megara" conda environment running python 2.7 but gives the following error when used under the "megara" conda environment in python 3.6. The error is: 

(megara3) bash-3.2$ python insert_fibers_in_extension.py 0001252338-20170629-MEGARA-MegaraLcbImage.fits
  File "insert_fibers_in_extension.py", line 30
    print error
              ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(print error)?

sergiopasra, 2018-07-31:
Yes, this is a Python3 incompatibility

sergiopasra, 2018-07-31:
This file is not the version in master. From where did you get the file?

acasmor, 2018-07-31:
I have it! don´t remember from where...

sergiopasra, 2018-07-31:
This is an old version. The current one is under tools in the repository

gildepaz, 2018-07-31:
I have used the version in github/megaradrp/tools (right after updating everything) within the conda environment and I get the error:
(megara) bash-3.2$ python ./insert_fibers_in_extension.py 0001251666-20170626-MEGARA-MegaraLcbImage.fits 
Traceback (most recent call last):
  File "./insert_fibers_in_extension.py", line 50, in <module>
    fibers = create_fibers(args.slit)
  File "./insert_fibers_in_extension.py", line 21, in create_fibers
    default_hdr = StringIO(data)
TypeError: initial_value must be str or None, not bytes

If I try this outside the environment with /Users/agpaz/anaconda3/bin/python ./insert_fibers_in_extension.py 0001251666-20170626-MEGARA-MegaraLcbImage.fits
it gives the same error but IT WORKS if I use instead:
/Users/agpaz/anaconda2/bin/python ./insert_fibers_in_extension.py 0001251666-20170626-MEGARA-MegaraLcbImage.fits

closed 2018-07-31 by sergiopasra

gildepaz, 2018-07-31:
Is it OK to close an "issue" just because it was moved as part of a milestone?

gildepaz, 2018-07-31:
Sorry, I saw that it was closed as part of "aa959d: Decode data from StringIO (Python3)" before rising my question.

commit aa959d20 (2018-07-31): Decode data from StringIO (Python3), fixes #218