guaix-ucm/megaradrp #219: Allow degree of polynomial function in script fiber_flat

acasmor, 2018-07-31:
In the case of LR-I we obtain a collapse-smooth fibler-flat as can be seen in the picture. The green curve should follow the data in a better way. We think having some kind of a free parameter in the spline fit we could get a better green curve.
[image]

sergiopasra, 2018-07-31:
@acasmor Which recipe produces this plot?

acasmor, 2018-07-31:
MegaraFiberFlatImage

gildepaz, 2018-07-31:
I believe smoothing_window is not supported since MegaraFiberFlatImage now uses a low-order polynomial instead. I think Afri is asking for the order to this polynomial to be tuneable as a parameter of the task, if possible.

sergiopasra, 2018-07-31:
I may be wrong, but MegaraFiberFlatImage uses a Savitzky-Golay filter:

https://github.com/guaix-ucm/megaradrp/blob/3663b68cb6d38a8965e4ca1d18e50704f6375972/megaradrp/recipes/calibration/flat.py#L143-L147

gildepaz, 2018-07-31:
Before it did not fit anything, it just smoothed but after we show that produced high-frequency features in the fiberflat correction we moved to fit the curve. That fit I would think it has parameters, so it would be great to have more control on the parameters of the fit to allow for this kind of locally-flat blue-to-red response to be better fitted. Thx!

sergiopasra, 2018-07-31:
Oh, yes, is that part
https://github.com/guaix-ucm/megaradrp/blob/3663b68cb6d38a8965e4ca1d18e50704f6375972/megaradrp/recipes/calibration/flat.py#L153-L156

Degree is already 5. It can't be larger than that with `Univariatespline`. I have no solution for this, currently

sergiopasra, 2018-07-31:
BTW, the points we are fitting are stored in a file `collapsed.txt`. 

You can try to find a better way to fit them. Something that is not a pline, perhaps. Then, we could expose, not the degree, but the type of fitting (spline-3, spline-5, etc), such as Iraf does. This would be a greater change, though

gildepaz, 2018-07-31:
I would leave this open until we another type of fitting can be added or, alternatively, allow the user to choose between the new and the old procedure for MegaraFiberflat, i.e. smoothing (old) instead of fitting (new).