guaix-ucm/megaradrp #220: If reference_extinction=None do not correct for extinction

acasmor, 2018-08-30:
MegaraLcbStdStar

sergiopasra, 2018-09-28:
So, the status of this is as follows:

In `LCBImageRecipe` and `MOSImageRecipe`, `reference_extinction` is **optional**
In `LCBStandardRecipe` and `MOSStandardRecipe` `reference_extinction` is  **required**

In `LCBImageRecipe` and `MOSImageRecipe`, `reference_extinction` and `master_sensitivity` are independent, so (currently) to avoid any flux calibration you have to **disable both**

One way to disable these requirements is assign the value `None` to them in the observation file.
The JSON value `null` is translated to `None` in Python, so adding this should work:

```yaml
requirements:
  reference_extinction: null
  master_sensitivity: null
```

acasmor, 2018-10-22:
Sergio, I´ve done the test with reference_extinction: null and master_sensitivity: null and now the values are "none" but I get the following error:

.....
DEBUG: correct from Twilight in image uuid:f569565e-d617-11e8-9b5d-c4b301d98315
numina.recipes.megara INFO start sky subtraction
numina.recipes.megara INFO obtain fiber information
numina.recipes.megara DEBUG sky fibers are: [22, 23, 24, 25, 26, 27, 28, 57, 58, 59, 60, 61, 62, 63, 134, 135, 136, 137, 138, 139, 140, 267, 268, 269, 270, 271, 272, 273, 351, 352, 353, 354, 355, 356, 357, 484, 485, 486, 487, 488, 489, 490, 561, 562, 563, 564, 565, 566, 567, 596, 597, 598, 599, 600, 601, 602]
numina.recipes.megara INFO end sky subtraction
numina.recipes.megara INFO no flux calibration
numina.recipes.megara INFO start extinction correction
Traceback (most recent call last):
  File "/Users/acm/anaconda2/envs/megara3/bin/numina", line 11, in <module>
    load_entry_point('numina==0.18.dev0', 'console_scripts', 'numina')()
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/cli.py", line 144, in main
    args.command(args, extra_args)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/clirun.py", line 107, in mode_run_obsmode
    mode_run_common(args, extra_args, mode='obs')
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 58, in mode_run_common
    return mode_run_common_obs(args, extra_args)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/clirundal.py", line 226, in mode_run_common_obs
    workenv=workenv, logger_control=logger_control)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/baserun.py", line 172, in run_recipe
    run_recipe_timed(task, recipe, rinput)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/user/baserun.py", line 185, in run_recipe_timed
    result = recipe(rinput)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/numina-0.18.dev0-py3.6-macosx-10.7-x86_64.egg/numina/core/recipes.py", line 146, in __call__
    result = self.run(recipe_input)
  File "/Users/acm/anaconda2/envs/megara3/lib/python3.6/site-packages/megaradrp-0.8.dev0-py3.6.egg/megaradrp/recipes/scientific/lcb.py", line 95, in run
    extinc_interp = interp1d(rinput.reference_extinction[:, 0],
IndexError: too many indices for array

commit 4257b002 (2018-12-11): Allow None value for extinction table (fixes #220)
closed 2018-12-11 by sergiopasra