guaix-ucm/megaradrp #230: Error running megaradrp-overplot_traces and also numina

gildepaz, 2019-02-14:
I run numina and I get the following error. I first thought it was a problem specific of numina but I got the same error running megaradrp-overplot_traces:

(megara) bash-3.2$ megaradrp-overplot_traces --rawimage data/0001900806-20181231-MEGARA-MegaraArcCalibration.fits obsid4-Tracemap_LCB_MR-G__results/master_traces.json --global_offset 0.0
>>> File..: data/0001900806-20181231-MEGARA-MegaraArcCalibration.fits
>>> NAXIS1: 4196
>>> NAXIS2: 4212
2019-02-14 17:48:35.309 python[31756:1441105] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7ffdd3928810
2019-02-14 17:48:35.312 python[31756:1441105] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7ffdd3928810'
*** First throw call stack:
(
        0   CoreFoundation                      0x00007fff31b10ecd __exceptionPreprocess + 256
        1   libobjc.A.dylib                     0x00007fff5dbc8720 objc_exception_throw + 48
        2   CoreFoundation                      0x00007fff31b8e275 -[NSObject(NSObject) __retain_OA] + 0
        3   CoreFoundation                      0x00007fff31ab2b40 ___forwarding___ + 1486
        4   CoreFoundation                      0x00007fff31ab24e8 _CF_forwarding_prep_0 + 120
        5   libtk8.6.dylib                      0x0000000b1a049134 TkpInit + 324
        6   libtk8.6.dylib                      0x0000000b19fa10ce Initialize + 2622
        7   _tkinter.cpython-37m-darwin.so      0x0000000b19dcaa0f _tkinter_create + 1183
        8   python                              0x00000001059558b6 _PyMethodDef_RawFastCallKeywords + 230
        9   python                              0x0000000105a92ba2 call_function + 306
        10  python                              0x0000000105a90852 _PyEval_EvalFrameDefault + 46114
        11  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        12  python                              0x0000000105954587 _PyFunction_FastCallDict + 231
        13  python                              0x00000001059d6c31 slot_tp_init + 193
        14  python                              0x00000001059e0c01 type_call + 241
        15  python                              0x0000000105955283 _PyObject_FastCallKeywords + 179
        16  python                              0x0000000105a92c35 call_function + 453
        17  python                              0x0000000105a90946 _PyEval_EvalFrameDefault + 46358
        18  python                              0x0000000105955075 function_code_fastcall + 117
        19  python                              0x0000000105a92b27 call_function + 183
        20  python                              0x0000000105a90852 _PyEval_EvalFrameDefault + 46114
        21  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        22  python                              0x0000000105954587 _PyFunction_FastCallDict + 231
        23  python                              0x00000001059584a2 method_call + 130
        24  python                              0x0000000105955ef2 PyObject_Call + 130
        25  python                              0x0000000105a90a9d _PyEval_EvalFrameDefault + 46701
        26  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        27  python                              0x0000000105955783 _PyFunction_FastCallKeywords + 195
        28  python                              0x0000000105a92b27 call_function + 183
        29  python                              0x0000000105a90852 _PyEval_EvalFrameDefault + 46114
        30  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        31  python                              0x0000000105955783 _PyFunction_FastCallKeywords + 195
        32  python                              0x0000000105a92b27 call_function + 183
        33  python                              0x0000000105a90946 _PyEval_EvalFrameDefault + 46358
        34  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        35  python                              0x0000000105955783 _PyFunction_FastCallKeywords + 195
        36  python                              0x0000000105a92b27 call_function + 183
        37  python                              0x0000000105a90946 _PyEval_EvalFrameDefault + 46358
        38  python                              0x0000000105955075 function_code_fastcall + 117
        39  python                              0x0000000105a92b27 call_function + 183
        40  python                              0x0000000105a9088d _PyEval_EvalFrameDefault + 46173
        41  python                              0x0000000105a841fe _PyEval_EvalCodeWithName + 414
        42  python                              0x0000000105ae7760 PyRun_FileExFlags + 256
        43  python                              0x0000000105ae6bd7 PyRun_SimpleFileExFlags + 391
        44  python                              0x0000000105b147bf pymain_main + 9583
        45  python                              0x0000000105927bbd main + 125
        46  libdyld.dylib                       0x00007fff5ec96ed9 start + 1
        47  ???                                 0x0000000000000007 0x0 + 7
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

gildepaz, 2019-02-14:
Searching for a solution I found that adding the following two lines in megaradrp-overplot_traces works (and it does): 
import matplotlib
matplotlib.use('TkAgg')
I have tried to add those two lines also in numina-check_wlcalib and numina-wavecalib. Let's see how it goes (still running). 

BTW, this seems to be a MacOSX only problema.

gildepaz, 2019-02-14:
It did not work in the case of the wavelength calibration. I am worried that it won't work for any task with graphical display. Please, check it out! THANKS!

In the case of the wavelength calibration the error starts with:
2019-02-14 21:26:58.943 python[33818:1476403] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7ff4623a8dd0
2019-02-14 21:26:58.945 python[33818:1476403] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7ff4623a8dd0'
*** First throw call stack:
(        0   CoreFoundation                      0x00007fff31b10ecd __exceptionPreprocess + 256
        1   libobjc.A.dylib                     0x00007fff5dbc8720 objc_exception_throw + 48
        2   CoreFoundation                      0x00007fff31b8e275 -[NSObject(NSObject) __retain_OA] + 0
        3   CoreFoundation                      0x00007fff31ab2b40 ___forwarding___ + 1486
        4   CoreFoundation                      0x00007fff31ab24e8 _CF_forwarding_prep_0 + 120
        5   libtk8.6.dylib                      0x0000000b24b9b134 TkpInit + 324
        6   libtk8.6.dylib                      0x0000000b24af30ce Initialize + 2622
        7   _tkinter.cpython-37m-darwin.so      0x0000000b2491ca0f _tkinter_create + 1183
        8   python                              0x000000010f0048b6 _PyMethodDef_RawFastCallKeywords + 230
        9   python                              0x000000010f141ba2 call_function + 306

acasmor, 2019-02-14:
It works for me! just with numina and megaradrp updated!

gildepaz, 2019-02-15:
I did not do that because the developers told me to start with a new conda installation without using the development one, sad!

gildepaz, 2019-02-15:
It finally worked editing also numina and numina-ximplotxy and numina-ximshow in anaconda3/bin/ and anaconda3/envs/megara/bin but I could not fix updating with git pull & python setup.py build + install because that part did not work in my installation (see issue #231 )

sergiopasra, 2019-02-15:
This is a problem with the documentation of the installation of numina (it happens via conda or pip, the error message is just different).

The *correct* solution of this problem is to install in your environment the default matplotlib backend.

To find out which backend your installation is considering the default, run the following command

```
python -c "import matplotlib;print(matplotlib.get_backend())"
```
Then, you have to be sure that package that provides this backend is installed

gildepaz, 2019-02-15:
It works is you precede the commands with MPLBACKEND=TkAgg or use a $HOME/.matplotlib/matplotlibrc with only "backend: TkAgg" (or any other backend of your like - that is installed, of course)

sergiopasra, 2019-02-15:
So, after a bit of testing with @gildepaz and @acasmor, the behavior seems to be:

If you are in a Mac with conda and the package suporting you backend is installed:

- With matplotlib 2.2, no failures
- With matplotlib 3.0
    - If your backend is defined via .matplotlibrc, no failures
    - If your backend is defined via environment MPLBACKEND, no failures
    - If your backend is the obtained from matplotlib defaults, it fails

The failure happens only if your backend comes from the matplotlib defaults, even when the backend is the same (TkAgg)

So this will fail
python -c "import matplotlib.pyplot as plt;plt.plot([1,2], [1,2]);plt.show()"

and this won't
MPLBACKEND=TkAgg python -c "import matplotlib.pyplot as plt;plt.plot([1,2], [1,2]);plt.show()"

This seems a matplotlib3/conda bug

sergiopasra, 2019-02-20:
Seems related to this https://github.com/ContinuumIO/anaconda-issues/issues/10219

closed 2020-03-17 by gildepaz