guaix-ucm/megaradrp #231: "python setup.py build" fails

gildepaz, 2019-02-15:
In order to fix a problem in the graphical output for Macs, I did a standard (cookbook-based) megaradrp installation. That installation worked. Since the problem with the graphical output remained, I did "git pull" and "python setup.py build" for numina to update to the newest version and I obtained this error:

(megara) bash-3.2$ git pull
Already up to date.
(megara) bash-3.2$ python setup.py build
We do not have Cython, just using the generated files
running build
running build_py
running build_ext
building 'numina.array._combine' extension
gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/agpaz/anaconda3/envs/megara/include -arch x86_64 -I/Users/agpaz/anaconda3/envs/megara/include -arch x86_64 -I/Users/agpaz/anaconda3/envs/megara/lib/python3.7/site-packages/numpy/core/include -I/Users/agpaz/anaconda3/envs/megara/include/python3.7m -c src/combinemodule.cc -o build/temp.macosx-10.7-x86_64-3.7/src/combinemodule.o
warning: include path for stdlibc++ headers not found; pass '-std=libc++' on the
      command line to use the libc++ standard library instead
      [-Wstdlibcxx-not-found]
In file included from src/combinemodule.cc:24:
In file included from /Users/agpaz/anaconda3/envs/megara/lib/python3.7/site-packages/numpy/core/include/numpy/arrayobject.h:4:
In file included from /Users/agpaz/anaconda3/envs/megara/lib/python3.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18:
In file included from /Users/agpaz/anaconda3/envs/megara/lib/python3.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1823:
/Users/agpaz/anaconda3/envs/megara/lib/python3.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: 
      "Using deprecated NumPy API, disable it by "          "#defining
      NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
#warning "Using deprecated NumPy API, disable it by " \
 ^
In file included from src/combinemodule.cc:26:
In file included from src/nu_combine_methods.h:25:
src/nu_combine_defs.h:27:10: fatal error: 'cstddef' file not found
#include <cstddef>
         ^~~~~~~~~
2 warnings and 1 error generated.
error: command 'gcc' failed with exit status 1
(megara) bash-3.2$ pwd   
/Users/agpaz/Desktop/MEGARA/Software/github/numina

nicocardiel, 2019-02-15:
I have found the same problem after upgrading to macOS Mojave. Try defining the following environment variable:

$ export MACOSX_DEPLOYMENT_TARGET=10.9

sergiopasra, 2019-02-15:
This should be in the documentation of numina