guaix-ucm/megaradrp #234: Add parameter with the number of rings to be added up in "StandardStar" recipe

gildepaz, 2019-02-22:
In order to ensure that all the flux is added up (especially when the star is very bright and there is some cross-talk) it would be great to be able to add up more rings of spaxels around the pre-defined offset. The maximum acceptable number should be 35 (to allow for all spaxels to be summed up but without reaching the position of the sky bundles).

sergiopasra, 2019-02-25:
It seems this is already implemented in MegaraLcbStdStar and MegaraMosStdStar, with parameter `nrings`

sergiopasra, 2019-02-25:
I can add this parameter to the LCB adquisition recipe, where it is fixed to 3

gildepaz, 2019-02-25:
Indeed, nrings=13 works but still there seems to be flux missing. Another option is that the sky spectrum is contaminated. Is there a way to tell to the MegaraLcbStdStar not to subtract the sky for computing the master_sensitivity?

sergiopasra, 2019-02-25:
> Indeed, nrings=13 works but still there seems to be flux missing. Another option is that the sky spectrum is contaminated. Is there a way to tell to the MegaraLcbStdStar not to subtract the sky for computing the master_sensitivity?

No, sky subtraction is currently hardcoded

sergiopasra, 2019-02-25:
> Indeed, nrings=13 works but still there seems to be flux missing. Another option is that the sky spectrum is contaminated. Is there a way to tell to the MegaraLcbStdStar not to subtract the sky for computing the master_sensitivity?

Still, you can try the parameter `ignored_sky_bundles` and put there all the sky bundles...

gildepaz, 2019-02-25:
I have used ignored_sky_bundles: [93,94,95,96,97,98,99,100] and although it does not give any error it always gives the following output:
_numina.recipes.megara INFO start sky subtraction
numina.recipes.megara INFO obtain fiber information
numina.recipes.megara DEBUG sky fibers are: [22, 23, 24, 25, 26, 27, 28, 57, 58, 59, 60, 61, 62, 63, 134, 135, 136, 137, 138, 139, 140, 267, 268, 269, 270, 271, 272, 273, 351, 352, 353, 354, 355, 356, 357, 484, 485, 486, 487, 488, 489, 490, 561, 562, 563, 564, 565, 566, 567, 596, 597, 598, 599, 600, 601, 602]
numina.recipes.megara INFO end sky subtraction_

sergiopasra, 2019-02-25:
This was a long shot anyway.

Could you try leaving all the bundles but one? Is the output the same?

gildepaz, 2019-02-25:
Yes, same output. If that list should show the number of fibers being used then the requirement "ignored_sky_bundles" is not working. This is the content of my 10_Standarstar_Box40.yaml file:
**id: 10-Box40Standardstar_LCB_MR-G_HR4554
mode: MegaraLcbStdStar
instrument: MEGARA
frames:
 - 0001900928-20190101-MEGARA-MegaraLcbStdStar.fits
requirements:
 reference_spectrum: mhr4554.dat 
 position: [-0.7495643878629911, -1.666524830716367]
 sigma_resolution: 40.
 extraction_offset: [2.0]
 nrings: 5
 ignored_sky_bundles: [93,94,95,96,97,98]
enabled: True**

commit c32dd68b (2019-02-26): Add parameter nrings to LCB acquisition (fixes #234)
closed 2019-03-04 by sergiopasra