guaix-ucm/megaradrp #235: Using the sky_rss.fits

gildepaz, 2019-02-25:
What can I do so the LcbImage recipe would use a given sky_rss.fits 2D spectrum (or a smoothed version of it) for the sky subtraction of the LCB image being processed instead of the one computed on its own?

gildepaz, 2019-02-26:
I would suggest that the LcbImage recipe would have a new requirement named "skyrss" that would indicate the name of a file in the data/ directory. If that requirement exists then it would simply subtract the 2D RSS from the reduced_rss.fits images to obtain the final_rss.fits file, instead of computing a single sky spectrum for all fibers. This has tremendous applications, especially when diffuse light is present and would avoid incorporating a full 2D sky subtraction from the recipe itself.

sergiopasra, 2020-05-18:
Closed with 670686a76b100ee8f338156e3359f95498d92acb

But I still consider this experimental

closed 2020-05-18 by sergiopasra