guaix-ucm/megaradrp #245: Update alignment angle

sergiopasra, 2019-03-25:
The alignment angle is defined only in the default argument of a function. Not the best place

https://github.com/sergiopasra/megaradrp-cube/issues/3

@gildepaz

gildepaz, 2019-03-25:
Could this be part of the model instead? Another option is that is a parameter that is associated with the instrument profile (currently ca3558e3-e50d-4bbc-86bd-da50a0998a48), because it could change if the rotator or the instrument has to be removed from the focus and put it back.

sergiopasra, 2019-03-25:
I'm writing this as a constant for the moment. I agree that it should go in some component of the instrument profile. I'm working on instrument profiles to make them more versatile

commit 21e52fa7 (2019-03-25): Add PLATESCALE and alignment angle to megaradrp.instrument (fixes #245)

* Add tests for pa_from_ipa
closed 2019-03-25 by sergiopasra