guaix-ucm/megaradrp #250: New release not working

gildepaz, 2019-06-28:
After following the instructions (and adding some to the MEGARA cookbook) I have failed to make the MEGARADRP work. The error is:

_(megara) bash-3.2$ numina
DEBUG: Numina simple recipe runner version 0.21
(megara) bash-3.2$ numina show-recipes
DEBUG: Numina simple recipe runner version 0.21
Problem loading MEGARA = megaradrp.loader:load_drp
Error is:  The 'lmfit' distribution was not found and is required by numina
(megara) bash-3.2$_

Someone forgot to include Lmfit in the list of things to be downloaded? I have run "conda activate megara" and that is not the problem.

gildepaz, 2019-06-28:
Sorry, I meant that I have run "conda update megara" and that is not the problem (either)

sergiopasra, 2019-06-29:
Fixed in https://github.com/conda-forge/numina-feedstock/pull/18

closed 2019-06-29 by sergiopasra