guaix-ucm/megaradrp #279: create_cube_from_rss returns nans for flux and velocity map

amanda-olmo, 2020-03-03:
While trying to draw contours with visualization, I have found that create_cube_from_rss returns a cube full of nans (line 421 in visualization). This happens when the contours are to be taken from the flux or velocity column. It does not happen when the continuum column is selected.

python -m megaradrp.visualization mrk324_f2_ha.fits --contour --contour-image mrk324_f2_ha.fits  --contour-image-column 17 -c 2 --hex-rel-size 1.26 
world coordinates in pixel
use column 2
world coordinates in pixel
use column 17
fiber coordinates in arcsec
fiber coordinates in arcsec
Estoy aqui
WARNING: FITSFixedWarning: CUNIT3 = / Units of coordinate increment and value 
a string value was expected. [astropy.wcs.wcs]
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1520: UserWarning: Warning: converting a masked element to nan.
  self.zmax = float(z.max())
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1521: UserWarning: Warning: converting a masked element to nan.
  self.zmin = float(z.min())
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1169: RuntimeWarning: invalid value encountered in less
  under = np.nonzero(lev < self.zmin)[0]
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1171: RuntimeWarning: invalid value encountered in greater
  over = np.nonzero(lev > self.zmax)[0]
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1200: RuntimeWarning: invalid value encountered in greater
  inside = (self.levels > self.zmin) & (self.levels < self.zmax)
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/matplotlib/contour.py:1200: RuntimeWarning: invalid value encountered in less
  inside = (self.levels > self.zmin) & (self.levels < self.zmax)
/Users/root2/opt/anaconda3/envs/megaradev/lib/python3.7/site-packages/megaradrp-0.10.dev0-py3.7.egg/megaradrp/visualization.py:423: UserWarning: No contour levels were found within the data range.
  mm = ax.contour(interp, transform=tt_d)
contour levels [nan]

sergiopasra, 2020-03-20:
@amanda-olmo Could you send me the image? Thanks

sergiopasra, 2020-04-14:
@amanda-olmo Amanda, the image itself has nan values. Cube doesn't mask or handle specially nan values as they are not expected to appear. nan values usually _spoil_ all the math operations unless handled specially. This means that if you image has nan values, the contours will be nan also.

My question is, the nan values represent something, missing data for example, or are they simply the result of a bad math operation, like 0 / 0? If it is the later, then this is not a bug

amanda-olmo, 2020-04-24:
The input file for the contours (mrk324_f2_ha.fits) is the output of the program analyse_rss.py which performs a fitting of the emission lines. In order to do the fit in a spaxel, the line must be above a certain signal to noise ratio. If the fit is not done, the program outputs a nan in almost every channel for that spaxel. So, the nan values represent missing data. 

I am not working with visualization.py anymore. I am using visualization_mine.py from Armando. The input of the program is the same as visualization.py, but it does not have this problem with the nan values. It can draw the contours.

sergiopasra, 2020-04-28:
Well, I can review the solution in @gildepaz program and see if it is feasible to use it in `visualization`