guaix-ucm/megaradrp #280: megaradrp-overplot_traces.py not backward compatible

gildepaz, 2020-03-17:
megaradrp-overplot_traces.py is not compatible with traces generated during the MEGARA commissioning. Date-obs field, for example, seems to be mandatory and it is not included in the old master_traces.json files. I have included that info by hand but I think megaradrp-overplot_traces.py should be able to handle master_traces.json files that lack that info, especially until the offsets are computed automatically based on temperature and date.

sergiopasra, 2020-03-20:
If the missing key is `date_obs` and the file has `insconf_uuid`, the program should work, so I'm going to fix it. 
Other missing keys can be more problematic.

sergiopasra, 2020-03-20:
Closed via https://github.com/guaix-ucm/megaradrp/commit/9760c6344d817ee38f0b068b0a0f592f32323cfa

closed 2020-03-20 by sergiopasra