guaix-ucm/megaradrp #316: Wave Calibration numina run fail: generate_fwhm_image function.

luisgonzalezramirez, 2024-10-30:
Hi! I am using megaradrp last version to reduce data from stars using both HR-R LCB mode and LR-R LCB mode. For all my data in LR-R LCB mode, I made all reduction steps correctly without any issue.

When I work with the HR-R mode, until the Wave Calibration step, everything goes fine. But in this step, I got this log info in the console:
INFO: ----------------------------------------------------
INFO: Starting row 614, fibid 615
INFO: number of peaks (expected): [20]
INFO: number of peaks (found)...: 20
WARNING: Insufficient number of points for fit, nfit=0
WARNING: problem in row 614, fibid 615
INFO: ----------------------------------------------------
for all fibers, and then a IndexError in the final array line 624 of recipes/calibration/arc.py:
...
  File "/path/.../megara_env/lib64/python3.11/site-packages/megaradrp/recipes/calibration/arc.py", line 445, in calibrate_wl
    image = self.generate_fwhm_image(initial_data_wlcalib.contents)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/path/.../megara_env/lib64/python3.11/site-packages/megaradrp/recipes/calibration/arc.py", line 624, in generate_fwhm_image
    voronoi_points = numpy.array(final[:, [0, 1]]) (*)
                                 ~~~~~^^^^^^^^^^^
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
Command failed with error: Command 'numina run 4_WaveCalib_LCB_LR-R_N110321.yaml -r control.yaml' returned non-zero exit status 1.

All my .yaml are correctly written just as I do for LR-R mode, and the control.yaml is the same as the one shown in the Cookbook and the one I used in LR-R mode. Also, the .fits images for WaveCalibration I copy in the data directory are the ones with SPECLAMP ThNe as it should be, and the fits image in ds9 are aligned with the raw traces got in ModelMap step.

When I add a print in the final variable (*) I get an empty array []. I have been some days with this issue.

My WaveCalib.yaml is:
id: 4_WaveCalib_LCB_HR-R_N110321
mode: MegaraArcCalibration
instrument: MEGARA
frames:
 - 0002879128-20210311-MEGARA-MegaraArcCalibration.fits
 - 0002879129-20210311-MEGARA-MegaraArcCalibration.fits
 - 0002879130-20210311-MEGARA-MegaraArcCalibration.fits
requirements:
 extraction_offset: [0.0]
 store_pdf_with_refined_fits: 1
enabled: True
which follows the same structure I use for LR-R mode (I create them with a code automatically).

In summary, with the LR-R mode I did not get any issue in my data reduction but I get it with HR-R mode.

Hope you could help me.

Luis, from CAB, CSIC-INTA.

nicocardiel, 2024-10-30:
Dear Luis. It seems that there is something wrong with fibid 615.

Have you checked that all the traces have been properly computed?
For that purpose, you can use the procedure explained in the new MEGARA cookbook (available online): https://guaix-ucm.github.io/megaradrp-cookbook/healtraces.html

It some traces are wrong, you must fixed them before proceeding with the wavelength calibration step.

luisgonzalezramirez, 2024-10-31:
Hi Nicolás! 
First of all, thank you for the help!

I have redo the steps to identify bad traces and I plot the trace map result with megara overplot. As you can see, the fibers are all fine, shown in the figure below.
[image]

The problem is with the fit of spectral lines calibration in each fiber. It seems the arc.py process can not detect any spec. lines. I show in the next figure the number of peaks found in wavecal_iter_1.pdf, in the obsid_ wave calibration _work dir.
[image]

The info message shown in my last comment appears for all the fibers in the wave calibration step.
The only fiber I miss in my trace map is the 623, but is something I also encounter with my LR-R mode reduction.

nicocardiel, 2024-10-31:
Dear Luis. At this point, the easiest way to understand what is going on is that you facilitate us a *tar.gz with the relevant files so we can execute the pipeline by ourselves. If you can do that, please, place that *tar.gz where we can download it. We can continue this "conversation" directly through email (cardiel@ucm.es, sergiopr@fis.ucm.es).

luisgonzalezramirez, 2024-10-31:
Great! I have just sent you two an email from my cab.inta-csic.es account: lgonzalez@cab.inta-csic.es

Cheers.

closed 2024-11-04 by nicocardiel